dusty-nv/jetson-inference #1847: werbrtc server is displayed

zerosnsa, 2024-05-17:
How to mask the data display below the webrtc video stream?
I want to block the ones framed in the picture
[image]

dusty-nv, 2024-05-17:
Hi @zerosnsa, you can do two things:

* make your own HTML app and have it look however you want:
   * https://github.com/dusty-nv/jetson-inference/blob/master/docs/webrtc-html.md
   * https://github.com/dusty-nv/jetson-inference/tree/master/python/www/html

* edit the HTML source of the webrtc debug page (what you are viewing now) inside jetson-utils:
   * https://github.com/dusty-nv/jetson-utils/blob/master/network/WebRTCServer.cpp

zerosnsa, 2024-06-04:
Whether to modify and recompile after I modify edit the HTML source of the webrtc debug page

zerosnsa, 2024-06-04:
I was thinking I would put your document.getElementById(statsElement).innerHTML = statsOutput; After commenting whether the page will not output those

zerosnsa, 2024-06-04:
I'm thinking, I'm going to put document.getElementById(statsElement).innerHTML = statsOutput; After commenting whether the page will not output those