dusty-nv/jetson-inference #1851: WebRTC don't work incase of no Internet

manhntm3, 2024-05-24:
I'm building a server that runs on Orin and view it through a PC in the same LAN network. When I ran video-viewer with a file as an input and webrtc output with this command:
`./aarch64/bin/video-viewer file://helicopter.mp4 webrtc://@:8554/output`

When opening the browser on an Internet-connected PC port 8554, it's running okay.
When opening it on a no Internet PC browser(within the same LAN), it keeps connecting and spinning.

Here is the full log: 
```
root@ubuntu:~/Software/ai_server/build# ./aarch64/bin/video-viewer file:///root/Software/cc/Videos/helofast.mp4 webrtc://@:8554/output
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for /root/Software/cc/Videos/helofast.mp4
No EGL Display 
nvbufsurftransform: Could not get EGL display connection
nvbuf_utils: Could not get EGL display connection
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 1920x1080  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)4, profile=(string)high, width=(int)1920, height=(int)1080, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=/root/Software/cc/Videos/helofast.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///root/Software/cc/Videos/helofast.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///root/Software/cc/Videos/helofast.mp4
     - protocol:  file
     - location:  /root/Software/cc/Videos/helofast.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      1920
  -- height:     1080
  -- frameRate:  29.97
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
failed to find/open file /proc/device-tree/model
[gstreamer] gstEncoder -- detected board 'Jetson-AGX'
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 idrinterval=30 maxperf-enable=1 ! video/x-h264 ! rtph264pay config-interval=1 ! application/x-rtp,media=video,encoding-name=H264,clock-rate=90000,payload=96 ! tee name=videotee ! queue ! fakesink
[webrtc] WebRTC server started @ http://ubuntu:8554
[webrtc] WebRTC server thread running...
[webrtc] websocket route added /output
[video]  created gstEncoder from webrtc://@:8554/output
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: webrtc://@:8554/output
     - protocol:  webrtc
     - location:  0.0.0.0
     - port:      8554
  -- deviceType: ip
  -- ioType:     output
  -- codec:      H264
  -- codecType:  v4l2
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- latency     10
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[... 387 lines cut ...]
{"sdpMLineIndex":0,"candidate":"candidate:9 1 TCP 1010828031 192.168.68.165 58493 typ host tcptype passive"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:1 2 UDP 2013266430 fe80::fcdf:90e6:cf:f0eb 52027 typ host"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:2 2 TCP 1015023102 fe80::fcdf:90e6:cf:f0eb 9 typ host tcptype active"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:3 2 TCP 1010828798 fe80::fcdf:90e6:cf:f0eb 35481 typ host tcptype passive"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:4 2 UDP 2013266429 172.17.0.1 53462 typ host"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:5 2 TCP 1015022078 172.17.0.1 9 typ host tcptype active"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:6 2 TCP 1010827774 172.17.0.1 36895 typ host tcptype passive"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:7 2 UDP 2013266428 192.168.68.165 43358 typ host"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:8 2 TCP 1015022334 192.168.68.165 9 typ host tcptype active"}
(index):59 Incoming ICE (ws://192.168.68.165:8554/output)
{"sdpMLineIndex":0,"candidate":"candidate:9 2 TCP 1010828030 192.168.68.165 35981 typ host tcptype passive"}
(index):185 WebRTC connection state (ws://192.168.68.165:8554/output) disconnected
(index):185 WebRTC connection state (ws://192.168.68.165:8554/output) failed
```

I noticed in the case without Internet on the Client PC, there is no ICE candidate being sent to the server. So the function `onIceCandidate` is never been triggered in this case.
Do you have any thoughts about this?

dusty-nv, 2024-05-24:
Hi @manhntm3, see this related issue - looks like it needs the STUN servers disabled in the javascript + gstreamer:  https://github.com/dusty-nv/jetson-containers/issues/535

dusty-nv, 2024-06-14:
Update on this here:  https://github.com/dusty-nv/jetson-containers/issues/535#issuecomment-2168857604

closed 2024-11-24 by manhntm3