dusty-nv/jetson-inference #1859: RTSP Video Output Fails on Jetpack 6

hunterwn, 2024-06-03:
I ran the following Python script after compiling and installing jetson-utils on my Orin devkit:

```python
from jetson_utils import videoSource, videoOutput

options = {}
video_input='/dev/video0'

input_stream = videoSource(video_input, options=options)

video_output='rtsp://@:1234/output'
output_stream = videoOutput(video_output)

while 1:
    image = input_stream.Capture(format='rgb8', timeout=2500)
    output_stream.Render(image)
 ```
 
 The result is that the camera displays fine in the OpenGL window but the RTSP server does not start successfully. 
 
 Here is the full output of the script:
 ```
python3 ./test.py
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Logitech BRIO
[gstreamer] v4l2-proplist, device.path=(string)/dev/video2, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Logitech\ BRIO", v4l2.device.bus_info=(string)usb-3610000.usb-3.4.4, v4l2.device.version=(uint)331656, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: Logitech BRIO
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Logitech\ BRIO", v4l2.device.bus_info=(string)usb-3610000.usb-3.4.4, v4l2.device.version=(uint)331656, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 43 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)540, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)848, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)440, height=(int)440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [11] video/x-raw, format=(string)YUY2, width=(int)480, height=(int)270, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [12] video/x-raw, format=(string)YUY2, width=(int)340, height=(int)340, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [13] video/x-raw, format=(string)YUY2, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [14] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [15] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [16] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [17] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [18] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [19] video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [20] video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [21] video/x-raw, format=(string)NV12, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [22] video/x-raw, format=(string)NV12, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [23] image/jpeg, width=(int)4096, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [24] image/jpeg, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [25] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [26] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [27] image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [28] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 90/1, 60/1, 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [29] image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [30] image/jpeg, width=(int)960, height=(int)540, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [31] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [32] image/jpeg, width=(int)848, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [33] image/jpeg, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [34] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 120/1, 90/1, 60/1, 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [35] image/jpeg, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [36] image/jpeg, width=(int)480, height=(int)270, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [37] image/jpeg, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [38] image/jpeg, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
H264: Profile = 66 Level = 0 
NVMEDIA: Need to set EMC bandwidth : 376000 
NvVideo: bBlitMode is set to TRUE 
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message qos ==> encoder
[gstreamer] gstreamer mysource ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline1/GstAppSrc:mysource:
streaming stopped, reason not-linked (-1)
```

InDroKalena, 2024-07-16:
Is there any update on this?

YIFEI-MA, 2024-07-22:
I had the exact same issue, webrtc works, but rtsp not with Internal data stream error.
`[gstreamer] gstreamer mysource ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline2/GstAppSrc:mysource:
streaming stopped, reason not-linked (-1)`
With exact same options:
`output = videoOutput("webrtc://@:8554/output", options={'codec': 'h265', 'bitrate': 1000000})
output2 = videoOutput("rtsp://@:8555/output", options={'codec': 'h265', 'bitrate': 1000000})`

pliolis, 2024-08-31:
Same issue with any of utility tool (video-viewer, detectNet etc):
```
[gstreamer] gstreamer message qos ==> encoder
[gstreamer] gstreamer mysource ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline1/GstAppSrc:mysource:
streaming stopped, reason not-linked (-1)
```

dusty-nv, 2024-08-31:
Hi guys, sorry haven't had time to dig into this - you can find where the RTSP pipeline is constructed here:

https://github.com/dusty-nv/jetson-utils/blob/e6aa502c9c612e0e89e513b7e60c5f98a755820f/codec/gstEncoder.cpp#L415C24-L415C28

My guess is that there were changes in the GstRtspServer plugins?  The code that uses those is here:

https://github.com/dusty-nv/jetson-utils/blob/master/network/RTSPServer.cpp

pliolis, 2024-10-29:
Is there any update on this?

JIA-HONG-CHU, 2024-11-20:
Is there any update on this? tks

ana-54n705h, 2024-12-05:
`appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM)  ! nvv4l2h264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 idrinterval=30 maxperf-enable=1 ! video/x-h264 ! **h264parse ! queue** ! rtph264pay config-interval=2 name=pay0`

Adding `**h264parse ! queue** ` before rtph264pay has solved video output issue for me. Add this in ` jetson-inference/utils/codec/gstEncoder.cpp`

Bonitodelcapo, 2025-01-14:
> `appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 idrinterval=30 maxperf-enable=1 ! video/x-h264 ! queue ! rtph264pay config-interval=1 name=pay0`
> 
> Hi @dusty-nv , Adding `queue !` before rtph264pay has SOLVED video output issue for me. Add this in ` jetson-inference/utils/codec/gstEncoder.cpp` then rebuild and install

Hi Ana, can you point out exactly how do patch this fix? looking at jetson-inference/utils/codec/gstEncoder.cpp i could not find the gstreamer string as the one that you posted, rather i could multiple parts scattered around the file.. could you be more precise and point exactly where the fix should be applied? thanks !

ana-54n705h, 2025-01-16:
HI @Bonitodelcapo, find `ss << "rtph264pay";` in the jetson-inference/utils/codec/gstEncoder.cpp:402 file and replace that with `ss << "queue ! rtph264pay";`. You can check in the logs whether this is applying or not in the gstreamer pipeline.

Bonitodelcapo, 2025-01-16:
hi @ana-54n705h thank you for your reply. i made the changes and it is visible in the pipeline log

> [gstreamer] gstEncoder -- codec not specified, defaulting to H.264
> [gstreamer] gstEncoder -- detected board 'NVIDIA Orin NX Developer Kit'
> [gstreamer] gstEncoder -- pipeline launch string:
> [gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 idrinterval=30 maxperf-enable=1 ! video/x-h264 ! **queue ! rtph264pay** config-interval=1 name=pay0
> [rtsp]   waiting for RTSP server to start...
> [rtsp]   RTSP server started @ rtsp://orinNX:1234
> [rtsp]   RTSP route added /my_output @ rtsp://orinNX:1234
> [video]  created gstEncoder from rtsp://@:1234/my_output

but i still get the error...

> [gstreamer] gstreamer mysource ERROR Internal data stream error.
> [gstreamer] gstreamer Debugging info: gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline1/GstAppSrc:mysource:
> streaming stopped, reason not-linked (-1)
> [gstreamer] gstreamer queue0 ERROR Internal data stream error.
> [gstreamer] gstreamer Debugging info: gstqueue.c(988): gst_queue_handle_sink_event (): /GstPipeline:pipeline1/GstQueue:queue0:
> streaming stopped, reason not-linked (-1)

could you maybe post your full working log so that i can compare it? 
Thanks !

ana-54n705h, 2025-01-16:
Just ignore warning messages for now, and check rtsp output video is coming or not.

pliolis, 2025-02-06:
I’m still getting the error too.

JIA-HONG-CHU, 2025-03-31:
Hey, I fixed this error by recompile the jetson-utils, but you need to revise the gstEncoder.cpp in codec folder.
In gstEncoder.cpp add 

	if( mOptions.resource.protocol == "rtsp" )
		ss << "queue max-size-buffers=1000 max-size-time=200000000 max-size-bytes=50000000 ! ";

between 

	else if( mOptions.codec == videoOptions::CODEC_MJPEG )
		ss << "! image/jpeg ! ";

and

	if( mOptions.save.path.length() > 0 )
	{
		ss << "tee name=savetee savetee. ! queue ! ";
		
		if( !gst_build_filesink(mOptions.save, mOptions.codec, ss) )
			return false;

		ss << "savetee. ! queue ! ";
	}