dusty-nv/jetson-inference #1860: Wrong data and working directory

JvnB, 2024-06-04:
Hi, 

im following along using the[ Docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md)

In image dustynv/jetson-inference:r36.3.0 I'm finding I'm having to change the Dockerfile to:

`DOCKER_ROOT="/opt/jetson-inference"`

If not the /jetson-inference folder will still be in the container, but that is because of the data volume. The build container is in `"/opt/jetson-inference".`

Without this correction also the outcome of the imagenet scripts wont be in the linked data volume so you cant view the results outside the container.

dusty-nv, 2024-06-20:
Thanks @JvnB, hopefully this is resolved with https://github.com/dusty-nv/jetson-inference/issues/1865#issuecomment-2181259842

JoostdeK, 2024-06-20:
Cool, I'll try it later. I'm @JvnB with another account ;)