dusty-nv/jetson-inference #1861: "video-viewer" Timeout Error with e-CAM25_CUONX Camera on Jetson Orin NX

Phantom-00, 2024-06-04:
I am experiencing an issue with accessing my e-CAM25_CUONX camera using the video-viewer application on my Jetson Orin NX. While the camera works perfectly with gst-launch-1.0, I encounter a timeout error with video-viewer.

Steps I did:

  Connect the e-CAM25_CUONX camera to the Jetson Orin NX.

  Verify camera functionality using the following GStreamer pipeline, which works without any issues:
  gst-launch-1.0 nvv4l2camerasrc device=/dev/video0 ! \
  "video/x-raw(memory:NVMM), format=(string)UYVY, width=(int)1920, height=(int)1080" ! \
  nvvidconv ! "video/x-raw(memory:NVMM), format=(string)I420, width=(int)1920, height=(int)1080" ! \
  nv3dsink sync=false

  Attempt to access the camera using the video-viewer application:

  video-viewer /dev/video0

  Behavior:
  [gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
  Additional Information:

The camera specifications, including resolution and frame rate, are supported.
Attempting to run video-viewer with explicit parameters did not resolve the issue.

Any help to resolve this issue would be greatly appreciated.

andanders, 2024-07-15:
Chiming in. I'm having the same issue, albeit sturdecam20 and Orin Nano. Did you resolve your problem?

Phantom-00, 2024-07-15:
Yes, switching from 4-lane to 2-lane has solved the problem.

andanders, 2024-07-15:
Sorry... what does that mean?

dusty-nv, 2024-07-15:
@andanders I think they mean the cable that connects the MIPI CSI camera to the Jetson.  I haven't exhaustively tested jetson-inference/jetson-utils with 3rd-party CSI cameras, so YMMV if you need to adjust the GStreamer pipeline to get video out of it.

Phantom-00, 2024-07-16:
When you install the driver, you will be asked in the terminal whether you want to install 2 lane or 4 lane drivers. You must therefore reinstall the driver and select 2 lane.

andanders, 2024-07-17:
I reinstalled the drivers and was not prompted to select number of lanes. But I suspect that the drivers default to 2 lanes by looking at the install script and `dmesg`
> [   13.872095] Device No of MIPI lane configuration is 2