dusty-nv/jetson-inference #1865: Jetpack 5 vs Jetpack 6 Missing Files

ahuckphin, 2024-06-11:
I have no issues with running jetson-inference on my Orin Nano running Jetpack 5. However when running Jetpack 6, following the exact steps I used previously, I ran into issue of missing files after docker/run.sh. 

Jetpack 5 screenshot:
[image]

Jetpack 6 screenshot: 
[image]

I am able to restore the files per se by running the build commands. However, I am wondering if this difference is intentional and/or there's a better workaround. Cause even after building the files, experience on Jetpack 6 continues to be different than on Jetpack 5. For example, a usable image directory containing the sample images is in a different location. 

I also noticed that on Jetpack 5, after docker/run.sh, I am automatically in a jetson-inference directory. However on Jetpack 6, after docker/run.sh, I am in root directory and need to first change directory (cd) to a jetson-inference directory.

bluevisor, 2024-06-14:
docker/build.sh asks me to specify a tag, which tag should I use for L4T R36.3.0?

JoostdeK, 2024-06-19:
#1860

dusty-nv, 2024-06-20:
Ahh, okay sorry everyone - on JetPack 6, I have moved jetson-inference docker over to being built in jetson-containers (this is so that will automatically be built on top of the rest of the stack instead of relying on static base images, and is able to be used when building other containers on top)

The idea was still to be able to use jetson-inference's `docker/run.sh` script to run it - but there was that issue with the symlinks, which should now be fixed in these commits:

* https://github.com/dusty-nv/jetson-containers/commit/930e99dbead6c6d5d8c58c05a2c869eaa01848d3
* https://github.com/dusty-nv/jetson-inference/commit/e8361ae7f5f3651c4ff46295b193291a93d52735

I rebuilt and tested it again here, and the contents are now showing up properly under `/jetson-inference`.   When you start the container with `docker/run.sh` on JP6, it will show the working directory as `/opt/jetson-inference`, but these are linked (by convention, in jetson-containers I build packages under `/opt` as to not pollute the root directory)

I also confirmed you can run things like `detectnet.py images/peds_0.jpg images/test/peds_0.jpg`, and when the container was started with `docker/run.sh` you will see the images show up on your host under `jetson-inference/data/images`

I have re-built and pushed the updated images for JP6 to `dustynv/jetson-inference:r36.2.0` and `dustynv/jetson-inference:r36.3.0`, so do a docker pull on those and let me know if you still face the issue.

ahuckphin, 2024-06-25:
Issue appears to be resolved now. Thanks.