dusty-nv/jetson-inference #1866: Stream through RTP to another PC using the videoOutput function, and disabling the glWindow

4uSpock, 2024-06-17:
Hi there!

I'm trying to stream to my local PC from the Jetson Nano, using the builtin videoOutput function. I've specified:
`jetson_utils.videoOutput("rtp://10.0.0.38:1234")`
 To save RAM, I'd also like to disable the automatically opening window, titled: "NVIDIA Jetson". 

It seems as though, it doesn't open a stream and also opens the aforementioned window.

Would be happy if you have a tip :)

dusty-nv, 2024-06-17:
Hi @4uSpock, can you post the terminal log here of when you run it, so we can see what GStreamer pipeline it creates?

Also you should be able to disable the window with `jetson_utils.videoOutput("rtp://10.0.0.38:1234", argv=['--headless'])`

4uSpock, 2024-06-17:
First of all, Dustin (@dusty-nv) , thank you for your amazing work! Your guide over at YouTube, and the docker container saved my grad. project! 
Thank you very much!
I'm happy I got an opportunity to thank you in person :)!

Second, that's the log.

```
root@jet:/my-detection/Project# python3 test.py
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: HP 320 FHD Webcam
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"HP\ 320\ FHD\ Webcam", v4l2.device.bus_info=(string)usb-70090000.xusb-2, v4l2.device.version=(uint)264701, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 20 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)2592, height=(int)1944, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)10/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)15/1;
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)25/1;
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [10] image/jpeg, width=(int)2592, height=(int)1944, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [11] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [12] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [13] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [14] image/jpeg, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [15] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [16] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [17] image/jpeg, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [18] image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] [19] image/jpeg, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1, 5/1 };
[gstreamer] gstCamera -- selected device profile:  codec=MJPEG format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720 ! jpegdec name=decoder ! video/x-raw ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: default
  -- ioType:     input
  -- codec:      MJPEG
  -- codecType:  cpu
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 ! video/x-h264 ! rtph264pay config-interval=1 ! udpsink host=10.0.0.38 port=1234 auto-multicast=true
[video]  created gstEncoder from rtp://10.0.0.38:1234
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://10.0.0.38:1234
     - protocol:  rtp
     - location:  10.0.0.38
     - port:      1234
[... cut ...]
H264: Profile = 66, Level = 40
NVMEDIA_ENC: bBlitMode is set to TRUE
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message stream-start ==> pipeline1
[gstreamer] gstreamer changed state from READY to PAUSED ==> udpsink0
[gstreamer] gstreamer message async-done ==> pipeline1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> udpsink0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline1
```

dusty-nv, 2024-06-17:
Oh cool, yes happy to help!   So yes, it appears to be creating the RTP output stream with H264 as intended.   Just for sanity, can you try this with video-viewer/video-viewer.py and see if you can receive the RTP stream?   Here are the ways I have tested it from the client side:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#viewing-rtp-remotely

VLC seems glitchy with GStreamer.   It might also be worth a quick test trying RTSP or WebRTC just to check if you can view those.

4uSpock, 2024-06-17:
I checked, and although RTP stream didn't work,  WebRTC worked well! The RTP stream seems to be working and streaming, my PC isn't getting it. Maybe there's a problem with it.

dusty-nv, 2024-06-17:
Interestingly, WebRTC uses RTP underneath.  It basically sets up the RTP stream for you.

On the client side, are you using VLC?  I haven't found that to be the most reliable in combination with GStreamer unfortunately (seems to prefer streams originating from ffmpeg)

4uSpock, 2024-06-17:
I am indeed using VLC. I've tried to use the Linux terminal on Windows but to no avail. WebRTC turns to be also unreliable, reporting:
`[gstreamer] gstEncoder -- pipeline full, skipping frame ...`

Did you encounter a solution to use on Windows to access the RTP stream?

B.t.w, I've encountered an issue when importing the `jetson_inference` and `jetson_utils`, after torch, torchvision, and openCV libraries. The issue was that the output glDisplay couldn't be used.