dusty-nv/jetson-inference #1870: I cannot download official models for inference

SchumacherVonHeinz, 2024-07-01:
According to the guidebook, I found codes for object detection within only 10 lines, like

#inference.py in jetson-inference
import jetson.utils
import jetson.inference
input = jetson.utils.videoSource("csi://0")
output = jetson.utils.videoOutput("display://0")
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
while output.IsStreaming():
    img = input.Capture()
    detections = net.Detect(img)
    output.Render(img)
    output.SetStatus("Performance {:.0f}FPS".format(net.GetNetworkFPS()))`

 but after i run "python inference.py", the terminal showed that

[TRT]    native precisions detected for GPU:  FP32, FP16, INT8
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    could not find engine cache .1.1.8502.GPU.FP16.engine
[TRT]    cache file invalid, profiling network model on device GPU

error:  model file 'networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff' was not found.
        if loading a built-in model, maybe it wasn't downloaded before.

        Run the Model Downloader tool again and select it for download:

           $ cd <jetson-inference>/tools
           $ ./download-models.sh

[TRT]    detectNet -- failed to initialize.
Traceback (most recent call last):
  File "inference.py", line 5, in <module>
    net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
Exception: jetson.inference -- detectNet failed to load network
(I only pasted the error part)

and I chose the imageNet using AlexNet, but still in failure(almost the same), what should I do to deal with this error? Thank u for ur attention! :)

dusty-nv, 2024-07-01:
Here is the model mirror page:  https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618

SchumacherVonHeinz, 2024-07-01:
> Here is the model mirror page: https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618
> […](#)

OMG! Thank you for ur fast reply! I'll look it right now!