dusty-nv/jetson-inference #1871: video-viewer --input-rate parameter not applied

evrdub, 2024-07-04:
Hello @dusty-nv,

I am trying to implement webrtc with a jetson orin on a custom board with on-semi MIPI cameras, and it works pretty well most of the time.

The command I use is :
```sh
video-viewer csi://0 webrtc://@:8555/output --input-width=640 --input-height=360 --input-rate=15  --stun-server disabled --headless
```
The other parameters are being applied correctly, but when I test I get the following result :
[image]
As you can see in the picture the command has `--input-rate=15` but `framesPerSecond: 48`, and sometimes it varries between 30 and 60, but is never 15 FPS.

Do you know how can I apply a custom framerate ? Maybe I did not write the value correctly as expected by video-viewer ?

Thank you in advance for your help.

dusty-nv, 2024-07-04:
Hi @evrdub , when the video-viewer starts, check the console log for the gstCamera GStreamer pipeline that gets created.  Then you can confirm that your `--input-rate` is actually being set in the pipeline:

https://github.com/dusty-nv/jetson-utils/blob/903fae30ecafaa5ecdfc8e34f4dec2025ad31778/camera/gstCamera.cpp#L161

I believe the mode needs supported by the camera though.  Otherwise, your own video-viewer or video-viewer.py script and throttle the `videoSource.Capture()` rate with calls to `sleep()` or similar.

evrdub, 2024-07-08:
Hello, thank you for your fast reply, indeed my camera does not support the 15fps mode, I think a way of preventing this for every camera could be to use `videorate` in the GStreamer pipeline, for example :
`videorate max_rate=15 ! video/x-raw(memory:NVMM),width=1920, height=1080,format=NV12 ! \`

closed 2024-07-30 by evrdub