dusty-nv/jetson-inference #1873: train action recognition model

tebie6, 2024-07-12:
Hi @dusty-nv

I want to train an action recognition model, but I'm not sure how to do it. Could you guide me? 

I would really appreciate your help.

Thank you very mush!

dusty-nv, 2024-07-12:
Hi @tebie6 these models came from Akash James' repo here - https://github.com/kingardor/Activity-Recognition-TensorRT , which in turn used base models from `Can Spatiotemporal 3D CNNs Retrace the History of 2D CNNs and ImageNet?`    which I believe that upstream repo has the training scripts.  So you would want to look into that process (I haven't trained my own of these)

tebie6, 2024-07-13:
Thank you for your quick response! It has been very helpful.