dusty-nv/jetson-inference #1881: can't detect far faces on custom model

raccococococo, 2024-07-30:
Hi, I'm new to the world of Jetson. I'm trying to create a custom model, and when I run the program "detectnet.py" with "--network=facedetect", I can see faces from 4-5 meters away from the camera. However, with my custom model, I can only see faces at 0.5 meters from the camera. Is this problem due to the minimal resolution of 300x300 in the model "mobilenet-v1-ssd-mp-0_675.pth"? Is there a model for training that uses photos at 80x80 or 40x40?

dusty-nv, 2024-07-30:
@raccococococo yes that is typically an issue with the model resolution, and SSD-Mobilenet can be trained at higher resolution (like 512x512), like here:

https://forums.developer.nvidia.com/t/how-train-jetson-inference-ssd512-model/168510/13?u=dusty_nv

(note that you no longer need to use different `res512` branch, that was merged into master)

raccococococo, 2024-07-31:
Can't I use another model like ResNet18 for this training? I have read that it can be used with 80x80 images

raccococococo, 2024-07-31:
I have tried using MobileNet V1 with 512x512 images and can always see faces at 2 meters. I need to see faces from farther away. The camera captures 2K images, so faces are seen well even from a distance, but the model can't keep up. Does anyone know of models that work with lower resolution?

raccococococo, 2024-08-07:
@dusty-nv any idea?