dusty-nv/jetson-inference #1883: How to use custom trained model in deepstream python apps

Vinaygoudasp7, 2024-08-06:
hi everyone can any one help me for useing custom trained model which is converted into onnx format in deepstream and is it possible to use yolo trained and converted into onnx format and then use it in deep stream python_apps plz help me with this

Vinaygoudasp7, 2024-08-07:
hi plz anyone can help

dusty-nv, 2024-08-07:
Hi @Vinaygoudasp7 !  Please post to the DeepStream SDK forum for assistance with DeepStream:  https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/deepstream-sdk/15

Here are a couple guides that I found quick:

* https://wiki.seeedstudio.com/YOLOv8-DeepStream-TRT-Jetson/
* https://docs.ultralytics.com/guides/deepstream-nvidia-jetson/
* https://docs.nvidia.com/tao/tao-toolkit/text/ds_tao/yolo_v4_ds.html