dusty-nv/jetson-inference #1884: Failed to create capturesession problem with jetson nano csi camera ( now i have no cameras available )

aliamrrr, 2024-08-06:
Hello , I recently had no space left and uninstalled docker image from dusty nv ( jetson-inference ) when I run my code I have this error .

Les vecteurs pointent dans la même direction générale.
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera – attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)720, height=(int)480, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video] created gstCamera from csi://0
gstCamera video options:
– URI: csi://0
- protocol: csi
- location: 0
– deviceType: csi
– ioType: input
– width: 720
– height: 480
– frameRate: 30
– numBuffers: 4
– zeroCopy: true
– flipMethod: rotate-180
[08/06/2024-14:30:25] [TRT] [I] [MemUsageChange] Init CUDA: CPU +225, GPU +0, now: CPU 298, GPU 3126 (MiB)
[08/06/2024-14:30:25] [TRT] [I] Loaded engine size: 21 MiB
[08/06/2024-14:30:27] [TRT] [I] [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +156, GPU +144, now: CPU 461, GPU 3295 (MiB)
[08/06/2024-14:30:28] [TRT] [I] [MemUsageChange] Init cuDNN: CPU +241, GPU +178, now: CPU 702, GPU 3473 (MiB)
[08/06/2024-14:30:28] [TRT] [I] [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +20, now: CPU 0, GPU 20 (MiB)
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer message stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:740 No cameras available
[gstreamer] gstCamera – end of stream (EOS)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstreamer pipeline0 recieved EOS signal…
[gstreamer] gstCamera::Capture() – a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera – stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera – pipeline stopped

Is it related to the doker because when I reinstall its not working . Please i need a solution !
working with jetson nano

dusty-nv, 2024-08-06:
> * nvarguscamerasrc0
>   Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:740 No cameras available

Hi @aliamrrr , it cannot find any MIPI CSI camera attached - are you in fact trying to use CSI camera?   It will try that by default, unless you specify V4L2, video stream, video file, or directory of images (see [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md))

Are you now running this outside of container, or inside container still?    When using MIPI CSI camera, it is always a good idea to try `nvgstcapture-1.0` program first.  This will confirm that you MIPI CSI camera is indeed connected and functioning properly, then you can try it in jetson-inference.

aliamrrr, 2024-08-07:
Thanks for the anwer 
Yes m using csi camera
I’m new to this so I will explain , in fact it was working , one day there was no space left in the device so it cant boot , I removed the jetson-inference dokcer image ( with docker rmi id …) , when It worked and I tested my code , it said failed to create capturesession(). I tried to re connect the camera and now its saying no cameras available . So this gst-launch-1.0 nvarguscamerasrc ! videoconvert ! fakesink is not working , ls /dev/video* says no such file or directory althought the camera is connected . When I re run the docker image I didnt work and I have no space left also

Please tell me the steps i can follow , ( when you say docker container are you talking about jetson-inference repo no ? )

dusty-nv, 2024-08-07:
Hi @aliamrrr, until you can get `nvgstcapture-1.0` to recognize your camera, jetson-inference won't find it either (regardless of using container or not).  You might check that the device is still operational.  If you still have issues debugging the camera itself, I recommend to post to the Jetson forums for hardware help:

https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70

Sorry to not be of more assistant with the camera hardware directly.