dusty-nv/jetson-inference #1891: How to train the ssd model using negative dataset

mepl432, 2024-09-13:
Hello,

All I read is that dataset for a single object detection in pascal voc format should have labels.txt file containing a single label which is to be identified and that labels.txt should not contain class 0 label of "BACKGROUND" in the dataset. I read that BACKGROUND Label will be generated in the folder where model is stored after training along with the class 1 label for actual object which is detected.

How then do I train the model with my negative samples dataset which does not contain the actual object which is detected in the positive samples with annotated bounding box. Negative samples are used to reduce false positives and there is a major need to train ssd model with it. 

@dusty-nv and community could you please guide. Thanks in advance

dusty-nv, 2024-09-13:
@mepl432 train_ssd.py will automatically add the `BACKGROUND` class and pull negative training samples from your dataset.  Otherwise you can modify the code as needed - mine is just a fork of the upstream repo here - https://github.com/qfgaohao/pytorch-ssd

mepl432, 2024-09-13:
Thank you for the prompt response @dusty-nv . When you say pull the negative samples, this is based on their Annotations xml file..correct?  so if my negativesample.jpg has corresponding negativesample.xml in Annotations folder as below , then it will be considered negative sample and the model will train on it using the 'BACKGROUND' label automatically?

```
<annotation>
  <folder>negative_samples</folder>
  <filename>negativesample.jpg</filename>
  <size>
    <width>640</width>
    <height>480</height>
    <depth>3</depth>
  </size>
  <segmented>0</segmented>
  <!-- No object tags -->
</annotation>

```

dusty-nv, 2024-09-16:
I think if the image has no bounding box labels, it gets dropped because it has no labels. But you can add a BACKGROUND object?  I would recommend digging into the code and modifying it to extract/apply the background examples like you want.

  
</annotation>