dusty-nv/jetson-inference #1895: In Plotly Dashboard, we are not able to add RTSP stream as Video input.

Vinayrraj, 2024-09-25:
In Plotly Dashboard, we are not able to add RTSP stream as Video input.

127.0.0.1 - - [25/Sep/2024 05:49:56] "GET /status HTTP/1.1" 200 -
127.0.0.1 - - [25/Sep/2024 05:49:56] "GET /resources HTTP/1.1" 200 -
adding stream /my_stream from source rtsp://admin:123456@192.168.29.81:554 with model ['ssd-mobilenet-v2']
[2024-09-25 05:49:56,495] ERROR in app: Exception on /streams [POST]
Traceback (most recent call last):
  File "/jetson-inference/python/www/dash/server/server.py", line 525, in _add_stream
    stream = Stream(self, args['name'], args['source'], args.get('models'))
  File "/jetson-inference/python/www/dash/server/stream.py", line 43, in __init__
    video_args += ['--input-codec=mjpeg', '--output-encoder=cpu']
TypeError: unsupported operand type(s) for +=: 'NoneType' and 'list'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/jetson-inference/python/www/dash/server/server.py", line 527, in _add_stream
    self.alert(f"Error creating stream {args['name']}", level="error", duration=0)
  File "/jetson-inference/python/www/dash/server/server.py", line 456, in alert
    Log.Error(f"[{self.name}] {text}")
NameError: name 'self' is not defined
127.0.0.1 - - [25/Sep/2024 05:49:56] "POST /streams HTTP/1.1" 500 -

Vinayrraj, 2024-09-25:
Same is the case with "Flask + REST",

akashshingha850, 2024-09-25:
hey!! what argument did you use for the plotly dashboard?? I am facing this issue.. 

jetson@ubuntu:/bme/jetson-inference/python/www/dash$ python3 app.py --detection=ssd-mobilenet-v2 --pose=resnet18-hand --action=resnet18-kinetics
usage: app.py [-h] [--host HOST] [--port PORT] [--ssl-key SSL_KEY]
[--ssl-cert SSL_CERT] [--resources RESOURCES]
app.py: error: unrecognized arguments: --detection=ssd-mobilenet-v2 --pose=resnet18-hand --action=resnet18-kinetics
jetson@ubuntu:/bme/jetson-inference/python/www/dash$