dusty-nv/jetson-inference #1898: install jetson-inference in Jetson orin nano with JetPack 6.1

Conley-Yang, 2024-10-07:
hi, I use Jetson orin nano with JetPack 6.1 and follow the step of [Building the Project from Source](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md#building-the-project-from-source)

I got the following error information when I executed **make -j$(nproc)**, is there any suggestion to install jetson-inference successfully? Thanks for your reading!

conley@conley-desktop:~/jetson-inference/build$ make -j$(nproc)
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaDraw.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFilterMode.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFont.cu.o
[  6%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaGrayscale.cu.o
[  7%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaNormalize.cu.o
[  8%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaOverlay.cu.o
[  9%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaPointCloud.cu.o
[  9%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaRGB.cu.o
[ 10%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaResize.cu.o
[ 11%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-affine.cu.o
[ 12%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-fisheye.cu.o
[ 12%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-intrinsic.cu.o
[ 13%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-NV12.cu.o
[ 14%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YUYV.cu.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/v4l2Camera.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o
[ 20%] Building CXX object utils/CMakeFiles/jetson-utils.dir/commandLine.cpp.o
[ 21%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o
[ 22%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaColorspace.cpp.o
[ 23%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaPointCloud.cpp.o
[ 23%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glBuffer.cpp.o
[ 24%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glCamera.cpp.o
[ 25%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
[ 26%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glEvents.cpp.o
[ 27%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glTexture.cpp.o
[ 27%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glWidget.cpp.o
[ 28%] Building CXX object utils/CMakeFiles/jetson-utils.dir/filesystem.cpp.o
[ 29%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageIO.cpp.o
[ 30%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageLoader.cpp.o
[ 31%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageWriter.cpp.o
[ 31%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devInput.cpp.o
[ 32%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devJoystick.cpp.o
[ 33%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devKeyboard.cpp.o
[ 34%] Building CXX object utils/CMakeFiles/jetson-utils.dir/logging.cpp.o
[ 34%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/IPv4.cpp.o
[ 35%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/IPv6.cpp.o
[ 36%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/Networking.cpp.o
[ 37%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/RTSPServer.cpp.o
[ 38%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/Socket.cpp.o
[ 38%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/URI.cpp.o
[ 39%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/WebRTCServer.cpp.o
[ 40%] Building CXX object utils/CMakeFiles/jetson-utils.dir/threads/Process.cpp.o
[ 41%] Building CXX object utils/CMakeFiles/jetson-utils.dir/threads/Thread.cpp.o
[ 42%] Building CXX object utils/CMakeFiles/jetson-utils.dir/timespec.cpp.o
[ 42%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOptions.cpp.o
[ 43%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOutput.cpp.o
[ 44%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoSource.cpp.o
[ 45%] Building CXX object utils/CMakeFiles/jetson-utils.dir/xml.cpp.o
[ 46%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
[ 46%] Built target jetson-utils
[ 47%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_tensorConvert.cu.o
[ 48%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o
[ 48%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_depthNet.cu.o
[ 49%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_backgroundNet.cu.o
[ 50%] Building CXX object utils/video/video-viewer/CMakeFiles/video-viewer.dir/video-viewer.cpp.o
[ 50%] Building CXX object utils/python/bindings_python_3.10/CMakeFiles/jetson-utils-python-310.dir/PyCUDA.cpp.o
[ 51%] Linking CXX executable ../../../aarch64/bin/video-viewer
[... 206 lines cut ...]
      |                                                                                                 %li
In file included from /home/conley/jetson-inference/build/aarch64/include/jetson-utils/cudaUtility.h:33,
                 from /home/conley/jetson-inference/c/tensorNet.h:32,
                 from /home/conley/jetson-inference/c/poseNet.h:27,
                 from /home/conley/jetson-inference/c/poseNet.cpp:23:
/home/conley/jetson-inference/build/aarch64/include/jetson-utils/logging.h:204:38: warning: format ‘%i’ expects argument of type ‘int’, but argument 8 has type ‘int64_t’ {aka ‘long int’} [-Wformat=]
  204 |         #define LOG_COLOR_RED        "\033[0;31m"
      |                                      ^
/home/conley/jetson-inference/build/aarch64/include/jetson-utils/logging.h:144:106: note: in definition of macro ‘GenericLogMessage’
  144 | #define GenericLogMessage(level, format, args...) if( level <= Log::GetLevel() ) fprintf(Log::GetFile(), format, ## args)
      |                                                                                                          ^~~~~~
/home/conley/jetson-inference/build/aarch64/include/jetson-utils/logging.h:150:79: note: in expansion of macro ‘LOG_COLOR_RED’
  150 | #define LogError(format, args...)               GenericLogMessage(Log::ERROR, LOG_COLOR_RED LOG_LEVEL_PREFIX_ERROR format LOG_COLOR_RESET, ## args)
      |                                                                               ^~~~~~~~~~~~~
/home/conley/jetson-inference/c/poseNet.cpp:160:17: note: in expansion of macro ‘LogError’
  160 |                 LogError("poseNet -- cmap dims (C=%i, H=%i, W=%i) don't match paf dims (C=%i, H=%i, W=%i)\n", DIMS_C(mOutputs[CMAP].dims), DIMS_H(mOutputs[CMAP].dims), DIMS_W(mOutputs[CMAP].dims), DIMS_C(mOutputs[PAF].dims), DIMS_H(mOutputs[PAF].dims), DIMS_W(mOutputs[PAF].dims));
      |                 ^~~~~~~~
[... cut ...]
      |                                                                                                       ~^
      |                                                                                                        |
      |                                                                                                        int
      |                                                                                                       %li
make[1]: *** [CMakeFiles/Makefile2:394: CMakeFiles/jetson-inference.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

ahmet-f-gumustas, 2024-10-09:
I got the same error . Maybe solve later. When I checked the repo, there is no release for R36.4 yet

keanrawr, 2024-10-11:
Same issue here.
I tried flashing JetPack 6.0, but the zip file seems to be corrupt, so I'm only able to use JetPack 6.1.
There is no docker image for JetPack 6.1, and I don't have a way to go flash the 6.0 image 🫠

Pelz17, 2024-11-07:
Will there be an image for r36.4.0?

rc5409, 2024-12-07:
Hi, I am trying to build the jetson inference, om jetson orin nano as well. 
I am getting the following error:
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o
In file included from /opt/jetson-inference/utils/codec/gstEncoder.cpp:24:
/opt/jetson-inference/utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /opt/jetson-inference/utils/codec/gstDecoder.cpp:24:
/opt/jetson-inference/utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /opt/jetson-inference/utils/codec/gstWebRTC.cpp:23:
/opt/jetson-inference/utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:254: utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:240: utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o] Error 1
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:282: utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:804: utils/CMakeFiles/jetson-utils.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c /tmp/build_jetson_inference.sh' returned a non-zero code: 2
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/tharunj29/Documents/new_try/jetson-inference/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/tharunj29/Documents/new_try/jetson-inference/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo DOCKER_BUILDKIT=0 docker build --network=host --tag jetson-inference:r35.5.0-jetson-inference --file /home/tharunj29/Documents/new_try/jetson-inference/jetson-containers/packages/ml/jetson-inference/Dockerfile --build-arg BASE_IMAGE=jetson-inference:r35.5.0-gstreamer /home/tharunj29/Documents/new_try/jetson-inference/jetson-containers/packages/ml/jetson-inference 2>&1 | tee /home/tharunj29/Documents/new_try/jetson-inference/jetson-containers/logs/20241207_003727/build/jetson-inference_r35.5.0-jetson-inference.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 2.

As far as I could find, the GstWebRTC is currently deprecated. How were you able to get past this?