dusty-nv/jetson-inference #1904: Building jetson-inference failing with JP6.0 on Jetson Orin

jasonthenderson, 2024-10-23:
I am trying to add jetson-inference to the ollama container and so I am going through the manual build steps within the ollama container before I update the Dockerfile to do this as part of the docker build process.  All the steps  in the Github documentation for building jetson-inference complete properly until I get to this step...any ideas?

root@orin:/jetson-inference/build# make -j$(nproc)
Consolidate compiler generated dependencies of target jetson-utils
[  1%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
[  4%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o
[  4%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o
[  4%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o
[  5%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o
[  5%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o
[  8%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glEvents.cpp.o
[  8%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glCamera.cpp.o
[  8%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
[  9%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glTexture.cpp.o
[  9%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glWidget.cpp.o
[ 10%] Building CXX object utils/CMakeFiles/jetson-utils.dir/filesystem.cpp.o
In file included from /jetson-inference/build/aarch64/include/jetson-utils/gstBufferManager.h:26,
                 from /jetson-inference/utils/camera/gstCamera.h:27,
                 from /jetson-inference/utils/camera/gstCamera.cpp:23:
/jetson-inference/build/aarch64/include/jetson-utils/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4324: utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /jetson-inference/utils/codec/gstEncoder.h:26,
                 from /jetson-inference/utils/codec/gstEncoder.cpp:23:
/jetson-inference/utils/codec/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
compilation terminated.
In file included from /jetson-inference/utils/codec/gstBufferManager.h:26,
                 from /jetson-inference/utils/codec/gstBufferManager.cpp:23:
/jetson-inference/utils/codec/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4380: utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o] Error 1
compilation terminated.
In file included from /jetson-inference/utils/codec/gstDecoder.h:26,
                 from /jetson-inference/utils/codec/gstDecoder.cpp:23:
/jetson-inference/utils/codec/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4352: utils/CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o] Error 1
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4366: utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o] Error 1
In file included from /jetson-inference/utils/codec/gstUtility.cpp:23:
/jetson-inference/utils/codec/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4394: utils/CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o] Error 1
In file included from /jetson-inference/utils/codec/gstWebRTC.h:27,
                 from /jetson-inference/utils/codec/gstWebRTC.cpp:23:
/jetson-inference/utils/codec/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
   30 | #include "NvInfer.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/build.make:4408: utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:664: utils/CMakeFiles/jetson-utils.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
root@orin:/jetson-inference/build#

sbow, 2025-01-19:
same issue for me

MrKangXiang, 2025-03-05:
你改了源文件？

cibi-p, 2025-11-08:
Try installing

```bash
sudo apt install nvidia-l4t-dla-compiler cuda tensorrt libnvinfer-dev
```

PlusV98, 2025-11-19:
> Try installing
> 
> sudo apt install nvidia-l4t-dla-compiler cuda tensorrt libnvinfer-dev

thx bro, really helpful