dusty-nv/jetson-inference #1908: nvcaffe_parser in TRT10

vijayreddysamula, 2024-11-06:
Hi @dusty-nv,
The TRT10 doesnt have nvcaffe_parser. but in the cMakeLists.txt still you have nvcaffe_parser. I have Installed with CUDA12.3 and TRT10.3 in x86_64.
I am getting always cannot find -lnvcaffe_parser: No such file or directory

if(CUDA_VERSION VERSION_LESS "12.6")
	target_link_libraries(jetson-inference nvcaffe_parser)
else()
    CUDA_ADD_CUBLAS_TO_TARGET(jetson-inference)
endif()

dusty-nv, 2024-11-06:
@vijayreddysamula it's set to 12.6 because that is when TRT10 began shipping in JetPack, but you can manually change your local CMakeLists.txt in that case, as through CMake I just get the CUDA_VERSION and not the TRT_VERSION to check against.

vijayreddysamula, 2024-11-06:
Thank you @dusty-nv for the quick reply.
I don't understand why there is still nvcaffe_parser in cMakeLists.txt.