dusty-nv/jetson-inference #1911: Issue in building jetson inference on Jetson nano

Shubhankargo, 2024-11-12:
Hi 

I am trying to build Jetson inference from source, following the instructions. but when i try to make, i am getting this error

shubhankar@shubhankar-jetson:~/jetson-inference/build$ make -j$(nproc)
[  1%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageLoader.cpp.o
[  1%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o
[  1%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageWriter.cpp.o
[  2%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devInput.cpp.o
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp: In function ‘cudaError_t cudaBayerToRGB(uint8_t*, uchar3*, size_t, size_t, imageFormat, cudaStream_t)’:
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:56:2: error: ‘NppStreamContext’ was not declared in this scope
  NppStreamContext nppStreamContext;
  ^~~~~~~~~~~~~~~~
[  2%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devJoystick.cpp.o
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:56:2: note: suggested alternative: ‘cuStreamCreate’
  NppStreamContext nppStreamContext;
  ^~~~~~~~~~~~~~~~
  cuStreamCreate
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:57:23: error: ‘nppStreamContext’ was not declared in this scope
  nppGetStreamContext(&nppStreamContext);
                       ^~~~~~~~~~~~~~~~
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:57:23: note: suggested alternative: ‘cuStreamCreate’
  nppGetStreamContext(&nppStreamContext);
                       ^~~~~~~~~~~~~~~~
                       cuStreamCreate
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:57:2: error: ‘nppGetStreamContext’ was not declared in this scope
  nppGetStreamContext(&nppStreamContext);
  ^~~~~~~~~~~~~~~~~~~
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:57:2: note: suggested alternative: ‘nppGetStreamNumSMs’
  nppGetStreamContext(&nppStreamContext);
  ^~~~~~~~~~~~~~~~~~~
  nppGetStreamNumSMs
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:60:27: error: ‘nppiCFAToRGB_8u_C1C3R_Ctx’ was not declared in this scope
  const NppStatus result = nppiCFAToRGB_8u_C1C3R_Ctx(input, width * sizeof(uint8_t), size, roi,
                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/shubhankar/jetson-inference/utils/cuda/cudaBayer.cpp:60:27: note: suggested alternative: ‘nppiCFAToRGB_8u_C1C3R’
  const NppStatus result = nppiCFAToRGB_8u_C1C3R_Ctx(input, width * sizeof(uint8_t), size, roi,
                           ^~~~~~~~~~~~~~~~~~~~~~~~~
                           nppiCFAToRGB_8u_C1C3R
[  3%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devKeyboard.cpp.o
utils/CMakeFiles/jetson-utils.dir/build.make:4427: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:703: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

I am running it on a Jetson Nano with jetpack 4.3 on ubuntu 18.04

Please help. I am not able to find a solution elsewhere.

Shus

sdxx-1, 2025-02-19:
> Hi
>
> I am trying to build Jetson inference from source, following the instructions. but when i try to make, i am getting this error
>
> shubhankar@shubhankar-jetson:~/jetson-inference/build$ make [...]

sdxx-1, 2025-02-19:
hello,have you found a solution?

Shubhankargo, 2025-02-19:
No I didn't. I downgraded to an older version of Jetpack which did solve this particular issue, but gave rise to a host of others for my application. 

Recently got an Orin nano super developer kit and everything works smooth now.

sdxx-1, 2025-02-19:
thank you so much!my English is poor.