dusty-nv/jetson-inference #1921: Unable to run `docker/run.sh` due to missing docker image for `r36.4.2` for jetson-inference

shaunliew, 2024-12-27:
Hello,

I just received **Jetson Orin Nano Super Kit** and  I'm currently trying to run the `docker/run.sh` script from the `jetson-inference` repository, but I'm encountering an issue because the Docker image for `r36.4.2` doesn't seem to exist. Here’s the output I get:

```
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R36.4.2
[sudo] password for orin_nano: 
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r36.4.2
DATA_VOLUME:      --volume /home/orin_nano/jetson-inference/data:/jetson-inference/data --volume /home/orin_nano/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/orin_nano/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/orin_nano/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/orin_nano/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/orin_nano/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
V4L2_DEVICES:     --device /dev/video0 
DISPLAY_DEVICE:   -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix 
Unable to find image 'dustynv/jetson-inference:r36.4.2' locally
docker: Error response from daemon: manifest for dustynv/jetson-inference:r36.4.2 not found: manifest unknown: manifest unknown.
See 'docker run --help'.
```

It seems that the image `dustynv/jetson-inference:r36.4.2` is missing from Docker Hub or is not available for my current L4T version (`r36.4.2`).

Can you confirm when the Docker image for `r36.4.2` will be available, or if there is any workaround to run the script in the meantime?

Thanks in advance!

RobotRos2001, 2024-12-27:
I also encountered the same problem, the new R36.4.2 does not have a corresponding image

dusty-nv, 2024-12-27:
Hello, try running it with:    `-c dustynv/jetson-inference:r36.3.0`

The r36.4 with TRT10 was not working with jetson-inference, so you can try previous image. I am updating Jetson AI Lab with newer way moving forward. I know that Ultralytics also got YoloV11 running on their Super Nano , that is a new model too.

shaunliew, 2024-12-27:
Thanks for the suggestion - I can confirm the command `docker/run.sh -c dustynv/jetson-inference:r36.3.0` works perfectly. Now im able to run jetson-inference docker container in jetpack 6.1 for orin nano super. Looking forward to your new Jetson AI Lab updates! On my end, I've been testing YOLOv11 on Super Nano with good results, though I'm currently troubleshooting CSI camera mounting with their Ultralytics docker container on Jetpack 6. Waiting to hear back from the Ultralytics team about this.

timthegod, 2025-02-09:
I can confirm the command docker/run.sh -c dustynv/jetson-inference:r36.3.0 works perfectly as well. My orin nano is on jetpack 6.2