dusty-nv/jetson-inference #1927: Unable to find image 'dustynv/jetson-inference:r36.4.3' locally

AaronSif, 2025-02-14:
I am running Jetpack 6.2.  Following the directions in the guide, when I enter docker /run.sh I get the following response rather than the prompt asking what model I want to use.  Any advise is appreciated.

ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R36.4.3
[sudo] password for nano1: 
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r36.4.3
DATA_VOLUME:      --volume /home/nano1/jetson-inference/data:/jetson-inference/data --volume /home/nano1/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/nano1/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/nano1/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/nano1/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/nano1/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
V4L2_DEVICES:     --device /dev/video0 
DISPLAY_DEVICE:   -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix 
Unable to find image 'dustynv/jetson-inference:r36.4.3' locally
docker: Error response from daemon: manifest for dustynv/jetson-inference:r36.4.3 not found: manifest unknown: manifest unknown.
See 'docker run --help'.

mapelli0, 2025-02-15:
Upvote, got the same issue

m986883511, 2025-02-24:
same error

mapelli0, 2025-03-17:
No updates on this issue? does anyone knows of some other examples that might be working then?

lopezfmr-dev, 2025-04-11:
This is not the solution but its a work around:
Edit the file "run.sh" with your favorite text editor
Add the following line at line 187  and before 200 

CONTAINER_IMAGE='dustynv/jetson-inference:r36.3.0'

that will pull the latest available container as of April 2025
Good luck!

diyu15, 2025-05-03:
> This is not the solution but its a work around: Edit the file "run.sh" with your favorite text editor Add the following line at line 187 and before 200
> 
> CONTAINER_IMAGE='dustynv/jetson-inference:r36.3.0'
> 
> that will pull the latest available container as of April 2025 Good luck!

thank you! this is work!

tadams79, 2025-11-05:
> I believe my issues are due to an updated docker for 36.4.3 not having been uploaded to the tutorial yet.  I was not aware that 36.4.3 was just uploaded Jan 15th, so maybe I am being impatient and will need to wait.  I have only had my Orin Nano for a couple weeks and it has been a little frustrating since I am using the latest version of everything and none of the tutorials I have found have been updated yet.

That is correct. That release hasn't been uploaded yet and, is the root cause of the issue.